Thor77/Blueproximity #68: Works or does not work?

gerroon, 2018-06-26:
HI

I am trying to enable and disable Mpd outputs using this app. Only function that seems to trigger anything is the promixty command. I cant get it to anything with the lock and unlock commands :( Maybe I am not getting it.

Thor77, 2018-06-26:
Hello there,
i'm very sorry to say that, but the current state of the code in this repository is not really usable, yet.
The backend is implemented nearly completely but the GUI is far from finished (work is taking place on the [gui-branch](https://github.com/Thor77/Blueproximity/tree/gui) for that).

Regarding your issue: Try running Blueproximity from a shell and look for log messages, if there's something wrong it should tell you. There are debug messages for (un)lock commands as well so you should be able to locate your problem more precisely.

gerroon, 2018-06-26:
@Thor77 

Thanks for the reply, I will try the command line part.

closed 2019-03-23 by Thor77