Thor77/Blueproximity #69: how i use it

jan10000000001, 2018-09-05:
i dont check from the read read me or other files how i install the complete programm oder how i start it

Thor77, 2018-09-06:
You have to install the bluez-dev package and it's python bindings (the name of those packages depends on your distro). In case your distribution doesn't package the python bindings, you can install them from [PyPi](https://pypi.org/project/PyBluez/).

Note this package currently only works with python 3, therefore you need the python 3 version of the bindings.

Once all requirements are installed, you can clone this repo, connect your bluetooth device to your computer (the process of doing that depends on the tools installed) and run the script by specifying the mac of your device:
```
$ git clone https://github.com/Thor77/Blueproximity
$ python3 -m blueproximity -m <mac of your connected bluetooth device>
```

jan10000000001, 2018-09-06:
now i can start it but what is the next step if kill my connection the programm  close it after some times no lock
ps mir fehlten ein haufen programme wo ich nicht wusste das ich sie brauchte

python-bluez 
bluez bluez-hcidump
libbluetooth-dev
libboost-python-dev 
libboost-thread-dev 
libglib2.0-dev

Thor77, 2018-09-06:
yeah, at the moment it doesn't support running as a daemon or with a gui/in autostart in any way,
so you have to start it manually and keep it running in some way (a systemd user-service could be suitable for that).
If your device disconnects, it should try to reconnect it. Is there any error message once the program closes?

Yeah, those are the ones I meant by python bindings, I guess they were all installed as a dependency of python-bluez/bluez, weren't they?

jan10000000001, 2018-09-06:
root@debian:/home/jan/Blueproximity# python3 -m blueproximity -m XX:XX:XX:XX:XX:XX,
[2018-09-06 23:39:22,409] DEBUG - Could connect on port 1
[2018-09-06 23:39:23,698] INFO - Starting daemon for "BluetoothDevice(mac=XX:XX:XX:XX:XX:XX, port=1, name=Boss, connected=True)"
[2018-09-06 23:39:23,710] DEBUG - Current distance: 0
[2018-09-06 23:40:23,777] DEBUG - Current distance: 1
[2018-09-06 23:40:23,778] INFO - Running command for new state lock
Traceback (most recent call last):
  File "/usr/lib/python3.5/runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "/usr/lib/python3.5/runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "/home/jan/Blueproximity/blueproximity/__main__.py", line 45, in <module>
    main()
  File "/home/jan/Blueproximity/blueproximity/__main__.py", line 39, in main
    worker.run()
  File "/home/jan/Blueproximity/blueproximity/worker.py", line 46, in run
    subprocess.run(state.command.split())
  File "/usr/lib/python3.5/subprocess.py", line 383, in run
    with Popen(*popenargs, **kwargs) as process:
  File "/usr/lib/python3.5/subprocess.py", line 676, in __init__
    restore_signals, start_new_session)
  File "/usr/lib/python3.5/subprocess.py", line 1282, in _execute_child
    raise child_exception_type(errno_num, err_msg)
FileNotFoundError: [Errno 2] No such file or directory: 'gnome-screensaver-command'
how i fixit  or you

Thor77, 2018-09-07:
Ah, that's a problem with the command it trie to run.
You have to specify a custom config with a custom command then:
```
[Lock]
distance=7
duration=7
command=<a command of your choice it runs once your device is out of range>
[Unlock]
distance=4
duration=1
command=<a command of your choice it runs once your device is in range>
```
Save those lines as test.cfg (replace the command) and start Blueproximity with `python -m blueproximity -m ... -c test.cfg`

Thor77, 2019-03-23:
Please reopen if your question is not answered.

closed 2019-03-23 by Thor77